FreeRTOS/FreeRTOS-Plus-TCP #1172: [Feature Request] Xilinx Zynq and Ultrascale functional differences in 4.x

JohnnyOpcode, 2024-07-25:
I'd like to migrate a MPSoC (Ultrascale) project to 4.x, but 

https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/main/source/portable/NetworkInterface/xilinx_ultrascale/NetworkInterface.c

is not keeping pace with

https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/main/source/portable/NetworkInterface/Zynq/NetworkInterface.c

htibosch, 2024-07-25:
Hello, thank you for your interest in FreeRTOS+TCP.

How do you mean that the Ultrascale driver is not keeping pace with the Zynq driver?

ActoryOu, 2024-07-26:
Hi @htibosch,
I think he means that the network interface of xilinx_ultrascale does not support multiple interface, which is introduced by TCP V4.

Hello @JohnnyOpcode,
We warmly welcome and appreciate contributions from the community to our repository. Your participation and code submissions are highly valued. Feel free to create PR once ready and we'll review it.

Thank you in advance for your support and involvement!

JohnnyOpcode, 2024-07-26:
> Hi @htibosch, I think he means that the network interface of xilinx_ultrascale does not support multiple interface, which is introduced by TCP V4.
>
> Hello @JohnnyOpcode, We warmly welcome and [...]

@ActoryOu I don't have the level of subject-matter expertise to code such a PR. I have my hands full with just the FreeRTOS robotic application (on MPSoC).

pete-pjb, 2024-07-26:
Hi @ActoryOu , @JohnnyOpcode and @htibosch ,

I did a port for this a number of months ago, but I have been too busy to contribute it...  Please find attached zip with source, if anyone has time to get it into the source tree please feel free!

[zus_gem.zip](https://github.com/user-attachments/files/16392705/zus_gem.zip)

I hope that is helpful!

Kind Regards,

Pete

htibosch, 2024-07-26:
Ah yes, now I understand it, Pete.
I will create a PR to upgrade the Ultrascale driver.
Thanks

htibosch, 2024-07-27:
I looked at the proposed new driver in detail. At the first sight it looks good.
I do have an Ultrascale, but at this moment, I don't have a budget (time) to review and test it.

closed 2024-11-07 by amazonKamath