FreeRTOS/FreeRTOS-Plus-TCP #1183: [BUG] Missing tick overflow protection on `u.xTCP.xLastActTime` and `u.xTCP.xLastAliveTime`

lzungri, 2024-09-07:
No tick overflow protection mechanism in place for `u.xTCP.xLastActTime` and `u.xTCP.xLastAliveTime`:
* https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/d70107967a14094994aecef68889b26e196ee6e0/source/FreeRTOS_TCP_Transmission.c#L969
* https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/d70107967a14094994aecef68889b26e196ee6e0/source/FreeRTOS_TCP_State_Handling.c#L183

Consider replacing raw arithmetic on the tick count with the safer `xTaskCheckForTimeOut`.

htibosch, 2024-09-07:
Hello @lzungri , thank you for reporting this, but I don't think that there is a problem with mentioned code.

~~~
   TickType_t uxAge = xTaskGetTickCount() - pxSocket->u.xTCP.xLastActTime;
~~~
Both `xTaskGetTickCount()` and `xLastActTime` have the type `TickType_t`, which is an `uint32_t` or bigger.

Suppose that `xLastActTime` has the value `0xFFFFFFF0`, and now the clocktick has overflown to `0x00000002`. Is that the overflow you mean? `uxAge` will become `0x00000012`, thanks to the magic of two's complement.

If you feel that I miss the point, please give an example in which things might go wrong.

lzungri, 2024-09-07:
@htibosch Thanks for looking into that.

> Suppose that xLastActTime has the value 0xFFFFFFF0, and now the clocktick has overflown to 0x00000002. Is that the overflow you mean? uxAge will become 0x00000012, thanks to the magic of two's complement.

When only one tick overflow has occurred, the result will be correct. I believe the problem will arise when more than 1 tick overflow occurred during that evaluation period. In that case, `uxAge` will not hold the right time delta as it doesn't account for the number of overflows.
This issue may not be that problematic on 32/64 bits CPUs where `TickType_t` is big enough to make sure an overflow only happens after a year or so. However, on ports where `TickType_t` is defined as `uint16_t` or `uint8_t`, overflows will happen much more frequently and could lead to timing/accuracy problems. Having a high `configTICK_RATE_HZ` will make things worse.

`xTaskCheckForTimeOut/vTaskSetTimeOutState` internally keep track of the number of overflows and thus will not have this problem.

htibosch, 2024-09-08:
I do not think that the +TCP is being used on a "16-bit" CPU. At least I don't know of any such project.

lzungri, 2024-09-08:
> I do not think that the +TCP is being used on a "16-bit" CPU. At least I don't know of any such project.

Got it, please let me know if I should close this issue.

htibosch, 2024-09-08:
If nobody else has comments on this subject, you may close it.
Thanks.

closed 2024-09-09 by aggarg