FreeRTOS/FreeRTOS-Plus-TCP #1190: [BUG]Missing parameters and undefined macros

lightning51522, 2024-09-24:
I am currently porting TCP packets of version 4.2.2 Release. On ATSAM4E, I have found issues with missing parameters when calling functions in the network interface source file, as well as the use of an undefined macro。

1.The function `usGeneratedProtocolChecksum` called in line 422 of the `portable/NetworkInterface/ATSAM4E/NetworkInterface. c` file is missing a length parameter.
2.The macro `ipLOCAL_MAC_ADDRESS` used on line 315 of the same file has not been defined.

Hope these errors can be fixed

kar-rahul-aws, 2024-09-25:
Hi @lightning51522 
I have raised a PR for fixing the missing parameter issue in ATSAM4E port. Thanks for pointing out the issue. 
Regarding your other concern, [DriverSAM ](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/main/source/portable/NetworkInterface/DriverSAM) replaces the older ATSAM4E port. So, you can use this port in your project . 
You can take a look at this [PR](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/957) , and make similar changes to the ATSAM4E port and raise a PR. We will be happy to review it. 

Thanks.

jasonpcarroll, 2024-09-26:
Can this issue be closed?

kar-rahul-aws, 2024-10-03:
Closing this issue since the [PR] (https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/1192) has been merged. Please reopen this issue if you feel otherwise.

closed 2024-10-03 by kar-rahul-aws