FreeRTOS/FreeRTOS-Plus-TCP #1193: [BUG] ATSAME5x network interface not detecting ICMP packets correctly

cruxic, 2024-09-27:
The example network interface for ATSAME5x does not correctly detect ICMP packets. The codes checks the protocol byte but fails to also check the Ethernet frame type. This causes it to get false-positive matches for frames which happen to have an 0x01 byte at a certain offset.

The same bug exists in two places in the code: xATSAM5x_NetworkInterfaceOutput() and prvEMACDeferredInterruptHandlerTask(). I will be submitting a pull-request which fixes it.

jasonpcarroll, 2024-09-27:
Hi @cruxic,
Thank you for bringing this to our attention. I will review your PR.

htibosch, 2024-09-28:
Good catch, thank you!

In the driver for Zynq, the ICMP packets are [detected correctly](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/main/source/portable/NetworkInterface/Zynq/NetworkInterface.c#L381), both for IPv4 and for IPv6.

The MSP432 driver has [same problem](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/main/source/portable/NetworkInterface/ThirdParty/MSP432/NetworkInterface.c#L801) and needs correction too.

aggarg, 2024-10-01:
Thank you for your contribution!

closed 2024-10-01 by aggarg