FreeRTOS/FreeRTOS-Plus-TCP #1199: [BUG] Build failure with DNS cache disabled

jackwilsdon, 2024-10-28:
**Describe the bug**
```
#define ipconfigUSE_DNS_CACHE 0
```
Removes the `pcName` field from `ParseSet_t`:
https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/57c1967bb0d5a5dfb4af3e9ac75d81ea931763a3/source/include/FreeRTOS_DNS_Globals.h#L192-L196

But `pcName` is still used by `DNS_ReadNameField`:

https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/57c1967bb0d5a5dfb4af3e9ac75d81ea931763a3/source/FreeRTOS_DNS_Parser.c#L104

**Target**
- Development board: Raspberry Pi Pico
- Instruction Set Architecture: ARMv6-M
- IDE and version: N/A
- Toolchain and version: arm-none-eabi-gcc (15:13.2.rel1-2) 13.2.1 20231009

**Host**
- Host OS: Linux
- Version: Debian GNU/Linux trixie/sid

**To Reproduce**
* Set `ipconfigUSE_DNS_CACHE` to `0`
* Note that FreeRTOS-Plus-TCP no longer compiles

**Expected behavior**
FreeRTOS-Plus-TCP should compile with the DNS cache disabled.

archigup, 2024-10-28:
Hi thanks for bringing this up! We're looking into this.

AniruddhaKanhere, 2024-10-28:
Hello @jackwilsdon,

Thank you for bringing this to our attention. You are absolutely correct. This is a bug in how the `#if`s are used.

The issue here is that the function `DNS_ReadNameField` only gets invoked when the DNS cache is enabled. But, it gets compiled with or without DNS cache being enabled. We should add a `#if ( ipconfigUSE_DNS_CACHE == 1 )` around the function declaration and definition. We will raise a PR to fix it shortly.

Thanks again,
Aniruddha

AniruddhaKanhere, 2024-10-31:
Hello @jackwilsdon, the above PR https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/1201 is merged in. Can you verify that the fix works for you?

Thanks,
Aniruddha

ActoryOu, 2024-11-05:
We're closing this issue as the patch has been successfully merged. If you encounter any further related issues, please don't hesitate to reopen this thread.

closed 2024-11-05 by ActoryOu