FreeRTOS/FreeRTOS-Plus-TCP #1207: Xilinx MPSoC R5F FreeRTOS TCP application

LVS-AI, 2024-11-24:
It's unclear if the Zynq driver should be used for an R5F FreeRTOS TCP application on the Xilinx MPSoC because of the 32-bit nature of the cores. I wonder if possible to document a compatiability matrix for the various Xilinx hard and soft cores including MicroBlaze (original) and MicroBlaze-V (Risc-V) and the corresponding FreeRTOS TCP portable driver to use.

Application development around FreeRTOS TCP is greatly slowed when there are unknowns regarding the correct software components to pull together for a given hardware configuration. Expertise and knowledge sharing are key to building a larger FreeRTOS ecosystem.

ActoryOu, 2024-11-25:
Hi @LVS-AI,
We appreciate your insights on this matter. If you're open to it, we'd love to know if you'd be willing to contribute these valuable clarifications to the project. Alternatively, if you prefer, our team would be happy to assist in implementing these improvements. Would you be interested in contributing directly, or would you like us to help incorporate these changes?

Thank you.

LVS-AI, 2024-11-25:
> Hi @LVS-AI, We appreciate your insights on this matter. If you're open to it, we'd love to know if you'd be willing to contribute these valuable clarifications to the project. Alternatively, if you [...]

Absolutely. I'm not savvy enough in the TCP realm so I look forward to learning alot on how the stack is put together from the subject matter experts (Amazon AWS etc.). I have a modest collection of Xilinx evaluation boards (VC707, ZC702, ZCU104) so testing coverage is reasonbly good going forward.

Who there can core dump their Xilinx knowledge on me so I can sort this all out into a nice tidy matrix? I'll need to be mentored/directed how best to proceed (roadmap).

ActoryOu, 2024-11-26:
Hi @LVS-AI,
The FreeRTOS-Plus-TCP repository includes support for different Xilinx platforms:
1. The [xilinx_ultrascale](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/main/source/portable/NetworkInterface/xilinx_ultrascale) directory contains support for Xilinx UltraScale+ devices, which use 64-bit Cortex-A53 processors.
1. The [Zynq](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/main/source/portable/NetworkInterface/Zynq) directory provides support for Xilinx Zynq devices.

If you're using a 32-bit processor like the R5F, we recommend using the Zynq implementation.

To improve the clarity of this information for other users, we'd greatly appreciate your help in enhancing the documentation. Would you be willing to contribute by adding these details to the respective README files in each directory?

Thank you.

LVS-AI, 2024-11-26:
My first thought is to express the matrix as a reworking of the repo directories. Something like ..

Xilinx
  - Cortex-M
    - TEMAC
  - MicroBlaze
    - TEMAC
  - MicroBlaze-V
    - TEMAC
  - MPSoC
    - A53
    - R5F
  - Versal
    - MRMAC
  - Zynq
    - A9

TEMAC driver(s) don't exist yet as far as we can tell. R5F driver is possibly the same as the Zynq A9 (not confirmed working).

In our particular use-case we want to serve-up certain AI acceleration (PL region) via a socket interface (micro-service) in a FreeRTOS TCP application. There is not a lot of documentation or samples to aid in this consolidation of knowledge.

ActoryOu, 2024-11-27:
I completely agree that presenting this information in a matrix format would be incredibly helpful for users.

Unfortunately, our team currently has limited access to a wide range of boards for comprehensive testing. We've been able to thoroughly test and document the Zynq and Ultrascale platforms, which is why we have detailed information available for these.

Regarding other SOCs, we would be immensely grateful for any contributions from the community. Please feel free to consult our comprehensive guide, [Porting FreeRTOS-Plus-TCP](https://www.freertos.org/Documentation/03-Libraries/02-FreeRTOS-plus/02-FreeRTOS-plus-TCP/10-Porting/01-FreeRTOS_TCP_Porting), which provides detailed instructions and best practices for this process. This document should serve as an excellent reference to support your integration efforts.

Thank you

ActoryOu, 2024-11-28:
Yes, and you might notice that these two PRs are also contributed by community. We believe the contributors have thoroughly tested these changes, and they should be compatible with the MPSoC series. However, I want to be transparent and mention that our team hasn't personally verified the functionality on MPSoC series platforms. We greatly value community involvement and would be delighted to welcome your contributions to the repository as well. If you have any questions or need assistance with contributing, please don't hesitate to reach out.

BTW, we have a [FreeRTOS Forum](https://forums.freertos.org/). Feel free to post your issue if you met any porting issue while integration.

LVS-AI, 2024-11-28:
> Yes, and you might notice that these two PRs are also contributed by community. We believe the contributors have thoroughly tested these changes, and they should be compatible with the MPSoC series. However, I want to be transparent and mention that our team hasn't personally verified the functionality on MPSoC series platforms. We greatly value community involvement and would be delighted to welcome your contributions to the repository as well. If you have any questions or need assistance with contributing, please don't hesitate to reach out.
> 
> BTW, we have a [FreeRTOS Forum](https://forums.freertos.org/), which has broader audience and embedded engineers around the world. Feel free to post your issue if you met any porting issue while integration.

Absolutely. Building sub-communities focused on specific platforms takes time and patience. Good guidance and documentation supports more widespread use of FreeRTOS (and TCP). We are inspired to refine, improve and even innovate whereever we can.

ActoryOu, 2024-12-02:
As the question has been thoroughly addressed, we'll be closing this thread.

closed 2024-12-02 by ActoryOu

LVS-AI, 2024-12-04:
> As the question has been thoroughly addressed, we'll be closing this thread.

We are waiting for the Gen 2 Versal to become more readily available to order an evaluation board. Other than that we think we have enough to figure out and test everything else. Thank-you for your help.