FreeRTOS/FreeRTOS-Plus-TCP #1217: [BUG] Pointless String Length in prvCreateDNSMessage() Leads to Compiler Warning/Error

StefanBalt, 2025-01-20:
**Describe the bug**

GCC (with strict settings) raises the following error

` error: 'strncpy' specified bound depends on the length of the source argument [-Werror=stringop-overflow=]`

for the line below

https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/fbb668ac7786c6f4ea8f2a60bfa2ae63c0b43863/source/FreeRTOS_DNS.c#L1507

and it is not wrong. Why use `strncpy()` if we do not have a useful length parameter in this scope?

I see 3 possible fixes:

1.  Keep track of the destination buffer size after it is allocated in `prvGetPayloadBuffer()`
2. Use `strcpy()` again (revert change from 4c4223a)
3.  Use `memcpy()`

tony-josi-aws, 2025-01-21:
@StefanBalt, thanks for reporting this issue.

This change was done along with other strcpy replacements since its known that `pucUDPPayloadBuffer` is big enough to hold the `pcHostName` as its allocated with the size of `pcHostName` accounted for in the size calculation [here](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/main/source/FreeRTOS_DNS.c#L806).

Since its causing compiler warnings with strict settings, it makes sense to update it, probably back to `strcpy` or `memcpy`.

GCC version: `gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0` doesn't produce this warning with any of the `-Wstringop-overflow=` types [0 to 4]. Are you using any other compiler flags?

StefanBalt, 2025-01-21:
Any GCC >= 9.2 I tried raised a warning with `-Wall -O2` flags. Lower optimization levels do not seem to warn interestingly.

Example in godbolt: https://godbolt.org/z/qaGdMGMKe

ig15, 2025-01-30:
The issue has been fixed by the [PR](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/1218). Thanks @StefanBalt for bringing it up. Closing the issue.