FreeRTOS/FreeRTOS-Plus-TCP #1224: [BUG] Include What You Use

HTRamsey, 2025-02-12:
Don't know if this is of any value to you guys, so close if not. But here are the results of running IWYU.

[freertos_tcp_iwyu.txt](https://github.com/user-attachments/files/18771590/freertos_tcp_iwyu.txt)

tony-josi-aws, 2025-02-12:
@HTRamsey thanks for sharing this. 

It would be great to have it included in the Github CI as a PR check as well to maintain it.

HTRamsey, 2025-02-12:
I just added
```
if(CMAKE_C_COMPILER_ID STREQUAL "Clang")
    find_program(INCLUDE_WHAT_YOU_USE_PROGRAM include-what-you-use)
    if(INCLUDE_WHAT_YOU_USE_PROGRAM)
        set(CMAKE_C_INCLUDE_WHAT_YOU_USE ${INCLUDE_WHAT_YOU_USE_PROGRAM})
    endif()
endif()
```

tony-josi-aws, 2025-02-13:
I tried applying the fixes that were generated (both locally as well as [freertos_tcp_iwyu.txt](https://github.com/user-attachments/files/18771590/freertos_tcp_iwyu.txt)) using IWYU `fix_includes.py` tool:
`python3 ~/Tools/include-what-you-use/fix_includes.py < freertos_tcp_iwyu.txt`

It seems like the build fails after applying the fixes because of missing macro definitions, mostly from header files that depended on other header files included in the same source files, as far as I checked.

HTRamsey, 2025-02-13:
Might have to make the header files self contained using --check_also. There's also other stuff you can do in the [documentation](https://github.com/include-what-you-use/include-what-you-use/blob/master/docs/IWYUPragmas.md) to fix some issues. But without some complex workflow set up for this, it might just be better to use it as a reference only and avoid the auto-fix.