FreeRTOS/FreeRTOS-Plus-TCP #1234: I see that the code for SLAAC for IPV6 is empty. Is there a plan to support SLAAC?

Elsa-is-My-Muse, 2025-03-18:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

htibosch, 2025-03-18:
What makes you thing that SLACC processing is not included?

~~~c
    #if ( ipconfigUSE_RA != 0 )

/**
 * @brief Work on the RA/SLAAC processing.
 * @param[in] xDoReset: When true, the state-machine will be reset and initialised.
 * @param[in] pxEndPoint: The end-point for which the RA/SLAAC process should be done..
 */
        void vRAProcess( BaseType_t xDoReset,
                         NetworkEndPoint_t * pxEndPoint );
    #endif /* ( ipconfigUSE_RA != 0 ) */
~~~

In FreeRTOS+TCP, DHCP is implemented for both IPv4 and IPv6.

For IPv6, beside DHCP, there is also "RA", or router advertising. The complete name should be "RA/SLAAC".

To summarise it (roughly):
An IPv6 enabled router advertises itself and the properties of the IPv6 network.
Note that IPv6 network often has a ton of available IP addresses.
When using "RA/SLAAC", a starting device chooses a random IP-address, and it will try it out using Duplicate Address Detection. When the IP-address is in use, it will receive a reply.
After probing a certain number of times, using fixed timeouts, it will decide to keep using the IP-address.
After a cold- or warm reboot, the device will choose a new random IP/address.

For more information, see for instance: [wikipedia](https://en.wikipedia.org/wiki/IPv6_address#Stateless_address_autoconfiguration_(SLAAC))

htibosch, 2025-03-24:
Hello @Elsa-is-My-Muse, I assume that the issue is solved by now?
Thanks

tony-josi-aws, 2025-04-07:
@Elsa-is-My-Muse 

Closing this issue as there is support for `RA/SLAAC` as explained by @htibosch 
Please feel free to reopen or start a new issue if your concern is not solved.

closed 2025-04-07 by tony-josi-aws