FreeRTOS/FreeRTOS-Plus-TCP #1238: [BUG] FreeRTOS-Plus/Demo/FreeRTOS_Plus_TCP_Echo_Qemu_mps2 stops working

laroche, 2025-04-03:
**Describe the bug**
Running the demo app FreeRTOS-Plus/Demo/FreeRTOS_Plus_TCP_Echo_Qemu_mps2
stops working after recent commit 3436954. DHCP will not complete to receive an IP address.

Not sure if the TCP code needs again modifications or if the demo app needs some
adjustments.

Just for the record, with the following hack you can also get the demo app running again:

<pre>
--- a/source/FreeRTOS_IPv4.c
+++ b/source/FreeRTOS_IPv4.c
@@ -449,7 +449,7 @@ enum eFrameProcessingResult prvAllowIPPacketIPv4( const struct xIP_PACKET * cons
                 /* The endpoint is not up, and the destination MAC address of the
                  * packet is not matching the endpoint's MAC address. Drop the
                  * packet. */
-                eReturn = eReleaseBuffer;
+                /* eReturn = eReleaseBuffer; */
             }
             else
             {
</pre>

**Target**
- qemu

**Host**
- Debian 12 or Debian 13 (devel snapshot)

**Expected behavior**
DHCP should work for this FreeRTOS-TCP demo app.

**Additional context**
Thanks for FreeRTOS-TCP network source code.

tony-josi-aws, 2025-04-04:
Thanks @laroche for contributing to FreeRTOS+TCP by creating the bug report.

Seems like the DHCP offers received on Qemu are with broadcast MAC as the destination address, even though the Advertise packets are sent with BOOTP settings having broadcast disabled and are getting dropped; hence, DHCP is not able to complete.

We have created a PR: https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/pull/1240 to address the issue. If you wish, please feel free to check out the PR.

laroche, 2025-04-04:
Great to see a patch and this fast response.

Thanks a lot,

Florian La Roche

closed 2025-04-05 by rohitmadan07

tony-josi-aws, 2025-04-07:
Thanks @laroche 
If you got a chance to check the PR, did it work for you?

laroche, 2025-04-07:
Yes, this patch works for me.

tony-josi-aws, 2025-04-07:
Thank you for confirming.