FreeRTOS/FreeRTOS-Plus-TCP #1276: [BUG]

kzorer, 2025-07-16:
# Reopen: Zero‑copy RX driver fails to allocate buffers on STM32H7xxx due to padding logic

## Description
When building FreeRTOS+TCP with `ipconfigZERO_COPY_RX_DRIVER=1` on STM32H7xxx, `pucGetRXBuffer()` always returns `NULL` and triggers the `configASSERT()` in the zero‑copy path. The failure appears to stem from the way buffers are sliced up in `uxNetworkInterfaceAllocateRAMToBuffers()`:

```c
size_t uxNetworkInterfaceAllocateRAMToBuffers( NetworkBufferDescriptor_t pxNetworkBuffers[ ipconfigNUM_NETWORK_BUFFER_DESCRIPTORS ] )
{
    uint8_t * ucRAMBuffer = ucNetworkPackets;
    for( uint32_t ul = 0; ul < ipconfigNUM_NETWORK_BUFFER_DESCRIPTORS; ul++ )
    {
        pxNetworkBuffers[ ul ].pucEthernetBuffer = ucRAMBuffer + ipBUFFER_PADDING;
        *( ( unsigned * ) ucRAMBuffer ) = ( unsigned ) ( &( pxNetworkBuffers[ ul ] ) );
        ucRAMBuffer += ETH_RX_BUF_SIZE;
    }
    return( ETH_RX_BUF_SIZE - ipBUFFER_PADDING );
}
```
Because ipBUFFER_PADDING = 8 + ipconfigPACKET_FILLER_SIZE, each descriptor’s “usable” buffer pointer gets offset in a way that, combined with STM32H7 DMA/cache requirements, results in no valid buffers being handed out, so pucGetRXBuffer() always returns NULL.

Note: It appears that this issue affects version 4.2.4 in the LTS release—using the LTS version is broken.

## Steps to Reproduce
1. In FreeRTOSIPConfig.h, set:
```c
#define ipconfigZERO_COPY_RX_DRIVER 1
#define ipconfigPACKET_FILLER_SIZE 2
#define ipconfigBUFFER_PADDING 0
#define ETH_RX_BUF_SIZE 1536
#define ipconfigNUM_NETWORK_BUFFER_DESCRIPTORS  64
```
2. Place the ucNetworkPackets[] buffer array in DTCM or mark SRAM as non‑cacheable via the MPU.
3. Call FreeRTOS_IPInit(), then generate Ethernet traffic into the STM32H7 MAC.
4. Observe the assertion:
```c
pucBuffer = pucGetRXBuffer( ETH_RX_BUF_SIZE );
configASSERT( pucBuffer != NULL );  // << fails here
```
## Expected behavior
pucGetRXBuffer() should return a valid pointer to the received Ethernet frame buffer, and the zero‑copy path should hand it off to the HAL without triggering an assertion.

## Actual Behavior
pucGetRXBuffer() returns NULL immediately, causing the configASSERT() to fire and preventing any packets from being processed.

## Environment
MCU: STM32H7xxx (e.g. STM32H743ZI)
FreeRTOS+TCP: v4.2.4 (LTS release)

jasonpcarroll, 2025-07-16:
Hi @kzorer,
Would you want to open a PR fixing the issue?
Best,
Jason

htibosch, 2025-07-17:
`pucGetRXBuffer()` will return NULL when all network buffers have been taken by `pxGetNetworkBufferWithDescriptor()`.
This can be caused by a configuration problem, or by not releasing buffers.

What is the value of `ETH_RX_DESC_CNT` / `ETH_TX_DESC_CNT`? Can you trace the usage of buffers? How many time did `pucGetRXBuffer()` return a buffer?

The famous +2 offset and Ethernet:

In most Ethernet peripherals, a 32-bit aligned pointer +2 is passed to Ethernet DMA. The reason is that later 32-bit fields (IP/TCP/UDP/ICMP etc) get well aligned.

In other Ethernet peripherals, an aligned pointer is sent to DMA, and the first 2 bytes remain unused.

The FreeRTOS IP-stack assumes that 4-byte values are well aligned.

The header looks like this:
~~~
    intptr_t owner;         // A pointer to the struct owning this buffer
    uint32_t filler_32;
    uint16_t filler_16;
    uint8_t ethernet_dest[6];
    uint8_t ethernet_source[6];
    uint16_t frame_type;
/* 32-bit aligned /
    uint8_t ucVersionHeaderLength;        /**< The version field + internet header length 0 + 1 =  1 */
    uint8_t ucDifferentiatedServicesCode; /**< Differentiated services code point + ECN   1 + 1 =  2 */
    uint16_t usLength;                    /**< Entire Packet size, ex. Ethernet header.   2 + 2 =  4 */
~~~

kzorer, 2025-07-18:
Our Ethernet configuration is unchanged from before—only the FreeRTOS+TCP version was bumped from **v4.2.2** to **v4.2.4**.  Below are the relevant settings:

```c
/* ########################### Ethernet Configuration ######################### */
#define ETH_TX_DESC_CNT    16U   /* number of Ethernet Tx DMA descriptors */
#define ETH_RX_DESC_CNT     8U   /* number of Ethernet Rx DMA descriptors */
```

- **FreeRTOSIPConfig.h:** No values were modified after the upgrade.
- **Observed behavior:** Immediately after switching to v4.2.4, the very first call to pucGetRXBuffer() returns NULL under identical load conditions.

Could you please confirm whether any changes were introduced in v4.2.4 related to buffer management—such as descriptor counts, alignment offsets, or allocation logic—that might explain this issue?

htibosch, 2025-07-18:
Your wrote:
~~~c
/* ########################### Ethernet Configuration ######################### */
#define ETH_TX_DESC_CNT    16U   /* number of Ethernet Tx DMA descriptors */
#define ETH_RX_DESC_CNT     8U   /* number of Ethernet Rx DMA descriptors */
~~~
The following values are preferred:
~~~c
#define ETH_TX_DESC_CNT    2U
#define ETH_RX_DESC_CNT   12U
~~~
That will not solve the problem, but it is more efficient.
Transmission: when the DMA queue has 2 descriptors, there will rarely be a shortage. And if both descriptors are found occupied, the program will wait in `xSemaphoreTake()`:

~~~c
if( xSemaphoreTake( xTxDescSem, pdMS_TO_TICKS( niEMAC_DESCRIPTOR_WAIT_TIME_MS ) ) == pdFALSE )
{
    FreeRTOS_debug_printf( ( "xNetworkInterfaceOutput: No Descriptors Available\n" ) );
    break;
}
~~~
It is much more useful to have enough RX descriptors. Incoming packets will be discarded when all RX descriptors are found full.

I tend to tune the value for `ETH_RX_DESC_CNT`: in the [Legacy version](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/blob/19134e1dc5adce673fa2f19f032e9245d8fc44e4/source/portable/NetworkInterface/STM32/Legacy/STM32Hxx/NetworkInterface.c#L1125) it remembers the highest usage of RX descriptors.

The above probably won't solve your problem. I will download the 2 versions and see if I find a problematic change.

htibosch, 2025-07-19:
I think that here things go wrong:

~~~c
NetworkBufferDescriptor_t * pxGetNetworkBufferWithDescriptor( size_t xRequestedSizeBytes,
                                                              TickType_t xBlockTimeTicks )
{
    NetworkBufferDescriptor_t * pxReturn = NULL;
    BaseType_t xInvalid = pdFALSE;
    UBaseType_t uxCount;

    if( ( xNetworkBufferSemaphore != NULL ) &&
        ( xRequestedSizeBytes <= uxMaxNetworkInterfaceAllocatedSizeBytes ) )
    {
        'uxMaxNetworkInterfaceAllocatedSizeBytes' is set at 1526 bytes, which is the buffer - 
    }
    else
    {
        /* My program lands here */
        /* semaphore not or created, or requesting too many bytes,
		1536 in stead of the maximum 1526. */
    }
~~~

I propose to change the following in the function `xSTM32H_NetworkInterfaceInitialise()`:

~~~diff
 /* Assign Rx memory buffers to a DMA Rx descriptor */
 for( uxIndex = 0; uxIndex < ETH_RX_DESC_CNT; uxIndex++ )
 {
     uint8_t * pucBuffer;
 
     #if ( ipconfigZERO_COPY_RX_DRIVER != 0 )
     {
-        pucBuffer = pucGetRXBuffer( ETH_RX_BUF_SIZE );
+        pucBuffer = pucGetRXBuffer( ETH_RX_BUF_SIZE - ipBUFFER_PADDING );
         configASSERT( pucBuffer != NULL );
     }
     #else
     {
         pucBuffer = Rx_Buff[ uxIndex ];
     }
     #endif
~~~

In the function `uxNetworkInterfaceAllocateRAMToBuffers()`, we allocate memory for all network buffers.
Each block is 1536 (`ETH_RX_BUF_SIZE`) bytes, but the first 10 bytes (`ipBUFFER_PADDING`) are reserved for administration.
So the following function is enitrely correct:
~~~c
size_t uxNetworkInterfaceAllocateRAMToBuffers( NetworkBufferDescriptor_t pxNetworkBuffers[ ipconfigNUM_NETWORK_BUFFER_DESCRIPTORS ] )
{
    uint8_t * ucRAMBuffer = ucNetworkPackets;
    uint32_t ul;

    for( ul = 0; ul < ipconfigNUM_NETWORK_BUFFER_DESCRIPTORS; ul++ )
    {
        pxNetworkBuffers[ ul ].pucEthernetBuffer = ucRAMBuffer + ipBUFFER_PADDING;
        *( ( unsigned * ) ucRAMBuffer ) = ( unsigned ) ( &( pxNetworkBuffers[ ul ] ) );
        ucRAMBuffer += ETH_RX_BUF_SIZE;
    }

    return( ETH_RX_BUF_SIZE - ipBUFFER_PADDING );
}
/*-----------------------------------------------------------*/
~~~

kzorer, 2025-07-28:
Hi again @htibosch,

I found the problem: because of the missing call `ipconfigEMAC_WATCHDOG_TIMER();`, the hardware watchdog isn’t being serviced. As I recall, I added it in earlier versions. May I open a PR to restore it so that everything works correctly?

```c
...
            else
            {
                /* Something has changed to a Link Status, need re-check. */
                prvEthernetUpdateConfig( pdFALSE );
            }
        }
        ipconfigEMAC_WATCHDOG_TIMER();
    }
}
...

tony-josi-aws, 2025-09-10:
@kzorer is there anything preventing you from using the main branch?

ActoryOu, 2025-09-15:
Closing the issue since the PR #1285 is merged. Feel free to reopen if you have any further question.

closed 2025-09-15 by ActoryOu